Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #1: Intel Edison arduino edition + I2C Motor Driver + Base Shield V2 (

raess1, 2017-10-22:
(i2cdetect)
[image]

I2C motor driver is contact to I2C on the Base Shield V2
it does not detect the I2C Motor Driver.

any suggestions?

KillingJacky, 2017-11-01:
Do you have an Arduino board? Could you try scan the I2C motor driver under Arduino?
```
// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    http://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// 
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

#include <Wire.h>

void setup()
{
  Wire.begin();

  Serial.begin(9600);
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```

raess1, 2017-11-01:
Sure,
Here is the result:
[image]

Also tried with and Grove 6Axis accelerometer & Gyroscope v1.0
here is the result:
[image]

As you can see the I2C motor driver cant be find.

The Motor Driver works because I can run it with UPM.

lanselambor, 2018-04-02:
Have a look by this issue https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver/issues/1
The new firmware has fixed this problem.

closed 2018-04-09 by lanselambor