Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #2: I2C unstable

eddie4, 2017-11-01:
Hello,

Am using Grove I2C Motor Driver v1_3 to control  a few motors in combination with a raspberry pi.  When using the command i2cdetect -y 1. However when doing this the first time it will correctly list the address 0f. Doing this a second 3th or 100th time will result in no results. When pressing reset button on the motor driver you will be able to detect it once. 

root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- 0f
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Also there are 3 other devices connected to the I2C bus these are not detected at all when motor driver is connected but are when it's disconnected. Above issue occurred with and without other I2C devices connected.

Pillar1989, 2017-11-20:
@eddie4  Do you modify  Grove_I2C_Motor_Driver_v1_3 firmware ?

eddie4, 2017-11-22:
No,  I didn't touch the firmware. This is what i got out of the box. You weren't able to reproduce the issue?

lanselambor, 2018-04-02:
@eddie4 This problem has been fixed at the new firmware. Please refer to https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver/issues/1

eddie4, 2018-04-02:
AWESOME!!!!

closed 2018-04-09 by lanselambor