Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #5: function & document for stepping motor

akita11, 2018-11-16:
I've tried to use a two-phase stepping motor using StepperRun() function, and it fails.
I've checked the source code of this library, and found that StepperRun() have two arguments, and the second one specifies the phase of two or four, and default is four phase.
There are no descriptions on this second argument of this function and the matter on the phase. I think it should be described in the document.

lanselambor, 2018-11-19:
I think on both .cpp and .h files we have had comment for the function. Isn't that right? Do we need to add something?
```
    // Drive a stepper motor
    // _step: -1024~1024, when _step>0, stepper motor runs clockwise; when 
    // _step<0, stepper motor runs anticlockwise; when _step is 512, the stepper
    // motor will run a complete turn; if step is 1024, the stepper motor will
    // run 2 turns.
    // _type: 0 -> 4 phase stepper motor
    //        1 -> 2 phase stepper motor
    void StepperRun(int _step, int _type = 0); 
```

akita11, 2018-11-19:
Yes, they are described in .cpp and .h, but the user needs to read these source codes to understand it. I think it is better that they are described in API document in readme.md.

lanselambor, 2018-11-19:
@akita11 Got you now, thanks for your suggestion, I will update readme.md. In usage 3 "Drive a Stepper Motor".

closed 2018-11-28 by lanselambor