Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #6: Serial speed

tchapi, 2018-11-27:
Hello,

Why starting the Serial port in the `begin()` method ? It prevents any other parts of the code to use the Serial at a different speed, and locks the serial monitor (since there is a change of speed mid-program).

The library should not init `Serial`, but rather assume that it's present (just like in the DHT library for instance : https://github.com/Seeed-Studio/Grove_Temperature_And_Humidity_Sensor)

tchapi, 2018-11-27:
A correct fix would just need to remove this line : 
https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver_v1_3/blob/master/Grove_I2C_Motor_Driver.cpp#L46

akita11, 2018-11-30:
There are other lines that use Serial. I think the use of Serial should be selectable by user, such as using #define. It would be used for debugging, not for normal use. The error would be checked by the return value of functions.

lanselambor, 2018-12-03:
@tchapi @akita11 Yes, we only need to remove __Serial.begin(9600);__ in the source file.

tchapi, 2018-12-03:
That's exactly what I'm talking about @lanselambor 👍

tchapi, 2018-12-03:
Thanks a lot !

closed 2018-12-03 by tchapi