Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #10: Fast stop

jmaurel87, 2022-03-08:
Hello,

I have noticed that the motor stop immediately only when it rotates anti-clockwise. When it rotates clockwise it doesn't stop immediately.

Sincerely

lakshanthad, 2022-03-14:
Hello @jmaurel87,

Could you please share the code you are trying to run?

Also, could you please attach 2 videos, one to show motor stop on clockwise rotation and another to show motor stop on anticlockwise rotation?

Thank you.

jmaurel87, 2022-03-24:
Hello,

Ok this is an extract from the code and a video. At the beginning the motor stop immediately (closing) and in the second part (opening) the motor don't stop immediately. I have the same problem with other motors.

Thanks for your attention

`
#include <Grove_I2C_Motor_Driver.h>

#define PIN_SOCKETLED_2  2
#define MOTOR_I2C_ADDR 0x0f

void Fermeture() {
  Etat_tempo = false;
  while (digitalRead(8) == HIGH && (digitalRead(4) == LOW && digitalRead(6) == HIGH)) {
    Motor.speed(MOTOR1, -1*100);
    ClignotementLED();
  }
  Motor.stop(MOTOR1);
}

void Ouverture() {
  Etat_tempo = false;
  while (digitalRead(3) == LOW && digitalRead(7) == HIGH) {
    Motor.speed(MOTOR1, 1*100);
    ClignotementLED();
  }
  Motor.stop(MOTOR1);
}

https://user-images.githubusercontent.com/76267928/159897925-6c3ade57-d5c7-4377-b685-8f7b82b64395.mp4

`

lakshanthad, 2022-03-25:
Hello @jmaurel87,

Thank you very much for the detailed information.

Can I know the exact model of the motor you are using in the video? Also, you said you tried other motors as well. Are those motors also the same model as the one you are using in this video?

Thank you.

jmaurel87, 2022-03-25:
Hello @lakshanthad 

The model in the video :
Modelcraft - RB350600-0A101R
Operating voltage : 12V/DC
Speed under load : 9 rpm
Max. gear load : 18kg/cm
Steady gear load : 6 kg/cm

An other motor tried with the same problem :
MFA / COMO DRILLS
12-24 DC
457RE280/1

lakshanthad, 2022-03-28:
Thank you @jmaurel87,

Previously you shared only part of the code. Could you share the entire code, please?

Thank you.

jmaurel87, 2022-03-29:
Hello @lakshanthad 

I've tested with the example of the library....and I have the same problem. Maybe it's a problem of the grove module.

#include "Grove_I2C_Motor_Driver.h"

// default I2C address is 0x0f
#define I2C_ADDRESS 0x0f

void setup() {
  Motor.begin(I2C_ADDRESS);
}

void loop() {
  Motor.speed(MOTOR1, 100);
    delay(2000);
  Motor.stop(MOTOR1);
    delay(2000);
  Motor.speed(MOTOR1, -100);
    delay(2000);
  Motor.stop(MOTOR1);
    delay(2000);
}

// End of file

BUT I've found a solution : Run the motor in the opposite direction for 15 ms and then stop the motor.

s-light, 2024-06-20:
without checking myself:
i think maybe your problem is not the library - 
the backslash in the gears of your motor could cause an *after-run*

closed 2024-10-30 by Lesords