Seeed-Studio/Grove_I2C_Motor_Driver_v1_3 #13: is there a release planed for the near future?

s-light, 2024-06-20:
i just stumbled on the new *Stepper-Motor* Mode options - and then found that these are currently only available when you manually install the library (download, unzip, move to arduino library folder....)

would love to have these feature easier available ;-)

s-light, 2024-06-20:
as in https://github.com/Seeed-Studio/Grove_LCD_RGB_Backlight/issues/23
i checked if the basic formating is *ok* :
```bash
$ arduino-lint --library-manager update
Linting library in /home/stefan/mydata/github/arduino_libraries/Grove_I2C_Motor_Driver_v1_3
WARNING: No header file found matching library name (Grove_I2C_Motor_Driver_v1.3.h). Best practices are for primary     
         header filename to match library name.                                                                         
         See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format         
         (Rule LS008)                                                                                                   
WARNING: library.properties name value Grove I2C Motor Driver v1.3 is longer than the recommended length of 16          
         characters.                                                                                                    
         See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format         
         (Rule LP010)                                                                                                   
WARNING: library.properties name Grove I2C Motor Driver v1.3 contains spaces. Although supported, best practices is to  
         not use spaces.                                                                                                
         See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format         
         (Rule LP015)                                                                                                   
WARNING: The library.properties paragraph field repeats the sentence field. These are displayed together so redundancy  
         is not needed.                                                                                                 
         See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format         
         (Rule LP036)                                                                                                   

Linter results for project: 0 ERRORS, 4 WARNINGS

-------------------

Linting sketch in /home/stefan/mydata/github/arduino_libraries/Grove_I2C_Motor_Driver_v1_3/examples/dcmotor_test

Linter results for project: no errors or warnings

-------------------

Linting sketch in /home/stefan/mydata/github/arduino_libraries/Grove_I2C_Motor_Driver_v1_3/examples/dcmotor_test_withoutLibrary

Linter results for project: no errors or warnings

-------------------

Linting sketch in /home/stefan/mydata/github/arduino_libraries/Grove_I2C_Motor_Driver_v1_3/examples/steppermotor_test

Linter results for project: no errors or warnings

-------------------

Linter results for projects: 0 ERRORS, 4 WARNINGS

```

Lesords, 2024-11-04:
Sorry, I've been busy lately.

I just checked and the latest version has been released. You can search for Grove I2C Motor Driver in Arduino and download version 1.0.4.

Lesords, 2024-11-04:
I'm going to close this issue, feel free to re-open it if you have any other questions.

closed 2024-11-04 by Lesords